ros-industrial-attic/industrial_moveit #19: libconstrained_ik_plugin.so: undefined symbol

DAIChengkai, 2015-08-26:
I followed the tutorial to add constrained_ik to my moveit_config and it works fine with the demo.launch. But when I want to use the ik_solver to my own package(like calling setFromIK function), error occurs and it said that /home/xxx/Imitation/devel/lib/IKPlan/load_model: symbol lookup error: /home/xxx/Imitation/devel/lib//libconstrained_ik_plugin.so: undefined symbol: _ZN2tf12poseMsgToKDLERKN13geometry_msgs5Pose_ISaIvEEERN3KDL5FrameE  . Can someone told me how to do the right configuration in a new package when you want to use the constrained_ik_solver.

Levi-Armstrong, 2015-08-26:
@DAIChengkai, Have you created your own constrained_ik moveit plugin or are you using constrained_ik/ConstrainedIKPlugin?

Levi-Armstrong, 2015-08-26:
@DAIChengkai, It looks like tf_conversions is not included in the find_package function in the Industrial_moveit/constrained_ik/CMakelist.txt.

Levi-Armstrong, 2015-08-26:
@DAIChengkai, I was able to determine that the libconstrained_ik_plugin.so does have undefined symbols using 

> $ ldd -d -r libconstrained_ik_plugin.so

At the end of the output I found:

> undefined symbol:   _ZN2tf12poseMsgToKDLERKN13geometry_msgs5Pose_ISaIvEEERN3KDL5FrameE	(./libconstrained_ik_plugin.so)
undefined symbol: _ZN2tf12poseKDLToMsgERKN3KDL5FrameERN13geometry_msgs5Pose_ISaIvEEE	(./libconstrained_ik_plugin.so)

After adding the **tf_conversions** as described in my previous comment and recompiling the code. I then ran the *$ ldd -d -r libconstrained_ik_plugin.so* and it was no longer undefined. Based on the findings I would expect that this should resolve your issue. I will submit a pull request to fix the issue once you verify that it fixes the issue.

DAIChengkai, 2015-08-27:
@Levi-Armstrong problem solved! all works fine after adding the tf_conversions. Thanks a lot!

Levi-Armstrong, 2015-08-28:
Glade to see it solved the issue.

shaun-edwards, 2016-01-11:
@Levi-Armstrong, can this be closed?

Levi-Armstrong, 2016-01-12:
@shaun-edwards, yep

closed 2016-01-12 by Levi-Armstrong