ros-industrial-attic/industrial_moveit #22: STOMP ur_driver velocities

rcj86, 2015-11-13:
I'm testing the STOMP planner with an UR5 real robot. I replaced OMPL by STOMP and works fine in simulation mode, but when I try to move the real robot, it doesn't move because only send positions, not velocities and accelerations.

jrgnicho, 2015-11-13:
@rcj86 thanks for your feedback.  How did you verify that only positions are populated? Did you by any chance run the command below to see what is in the trajectory?
```
rostopic echo -n 1 /joint_path_command
```

rcj86, 2015-11-18:
Sorry for my late reply. I looked in the topic /follow_joint_trajectory/goal and only positions are send, velocities and accelerations are empty. But if I change to OMPL, the robot execute the trajectory and sends velocities and accelerations.

OMPL:
[image]

STOMP:
[image]

Also I run the command rostopic echo -n 1 /joint_path_command, but I only saw this

WARNING: topic [rostopic echo -n 1 /joint_path_command] does not appear to be published yet

jrgnicho, 2015-11-18:
@rcj86 
Make sure that the stomp_planning_pipeline.launch file in your package contains the **AddTimeParametrization** planning adapter as shown below:
```
  <arg name="planning_adapters" value="default_planner_request_adapters/AddTimeParameterization
                                       default_planner_request_adapters/FixWorkspaceBounds
                                       default_planner_request_adapters/FixStartStateBounds
                                       default_planner_request_adapters/FixStartStateCollision
                                       default_planner_request_adapters/FixStartStatePathConstraints" />
```
I verified that by adding this, the message in ** /follow_joint_trajectory/goal** topic gets it's accelerations and velocities populated.
This adapter isn't listed in the stomp_test moveit package but I'll work on adding it there.  
@shaun-edwards this topic may be of interest to you.

rcj86, 2015-11-19:
I added the AddTimeParametrization in the stomp_planning_pipeline.launch, and now I'm able to control the robot.

Thanks!!

closed 2015-11-19 by rcj86