ros-industrial-attic/industrial_moveit #25: Documentation and Cleanup

jrgnicho, 2016-01-07:
Documentation is missing, at the very least the steps to run any existing demos should be documented along with instructions for how to load STOMP and CLIK into moveit.  Furthermore, launch, unit tests and executables files in STOMP that are no longer part of the build should be removed as they make it confusing when one inspects the ros packages.

yijiangh, 2017-11-07:
I'm wondering if we can have a more detailed tutorial or guidance on how to configure STOMP on a different robot other than the existing KR210 demo?

I'd love to document it if someone can give me a brief description of the major steps and order.

Levi-Armstrong, 2017-11-08:
Have you taken a look at the [examples page](https://github.com/ros-industrial/industrial_moveit/blob/kinetic-devel/stomp_moveit/example_pages.dox)? It provides more detailed information about each of the parameters and plug-ins. You can generate the full documentation using the rosdoc_lite package.

jrgnicho, 2018-07-05:
This documentation content that's now available should be pushed to the wiki section.  @Levi-Armstrong any thoughts on how to best do this?