ros-industrial-attic/industrial_moveit #29: Stomp: Planning to a goal state equal to the start state generates an unnecessary movement

beatrizleon, 2016-10-25:
I am testing Stomp and I noticed that when I ask it to plan a movement to a goal state equal/similar to the start state it generates an unnecessary movement instead of remaining there. Here is a video of what I mean:

https://www.opentest.co/share/341626809aaf11e6b14b47baa7c14fe2

@jrgnicho Is there something in the config that I could modify to avoid that? or is this normal for stomp?

jrgnicho, 2016-10-25:
Thanks for pointing this out @beatrizleon , it's difficult to tell what's happening just from the video alone, do you have a repo that reproduces this issue that I can tests?

beatrizleon, 2016-10-25:
I haven't make any changes to moveit. I just test to set in rviz the start and goal state as current and try to plan it. Then I always get the unnecessary movement. I tested this using ompl and it doesn't happen. I guess it can be reproduced with any robot.

jrgnicho, 2016-10-25:
@beatrizleon my first guess is that you might need to tweak the [`stddev`](https://github.com/ros-industrial/industrial_moveit/blob/indigo-devel/stomp_test_kr210_moveit_config/config/stomp_config.yaml#L16) parameters.  The larger the values the greater the motion that that joint will make in order to explore the workspace.  You may have values that are too large

closed 2018-07-05 by jrgnicho