ros-industrial-attic/industrial_moveit #44: Kinetic release?

ipa-fxm, 2017-07-12:
Any plans to release this repo for Kinetic any time soon?
This is currently blocking an urgent release of our cob-repositories, in particular the release of `cob_moveit_bringup` and thus `cob_manipulation` because we are depending on `stomp_moveit` and `industrial_collision_detection` (see https://github.com/ipa320/cob_manipulation/blob/kinetic_dev/cob_moveit_bringup/package.xml#L17-L26).

@ipa-bfb @ipa-fmw FYI

Levi-Armstrong, 2017-07-12:
@jrgnicho, What is the current status of the kinetic release?

ipa-fxm, 2017-07-12:
it's never been released yet...I don't know about the functional status of the repo....it's the missing rosdep key that is blocking us...:wink:

jrgnicho, 2017-07-13:
I'm working on getting some [new features](https://github.com/jrgnicho/industrial_moveit/tree/update-stomp-plugins-ik-code) in there before releasing it.  I'll wrap that up and work on the release over the weekend.

ipa-fxm, 2017-07-13:
awesome, thanks!

closed 2018-07-06 by jrgnicho