ros-industrial-attic/industrial_moveit #51: Set TCP speed

mat-l, 2017-08-24:
Dear all,
Dear @gavanderhoorn 

we are working with ROS indigo right now using a Universal Robot UR10. I am wondering if there is the possibility to set a limit of the TCP speed. There are lot of applications where one needs to set a defined TCP speed, like 100 mm/s or other values.

Anyway, I just found the two scaling factors which are most like only regarding the joint speed:
```
      group.setMaxVelocityScalingFactor(0.25);
      group.setMaxAccelerationScalingFactor(0.25);
```
These scaling factors refer to the URDF files of a robot. In Universal Robot that means for example:
`velocity="3.2"/` 
Which unit is that? Is it only representing the joint speed in point to point movement or can I transfer this also to Cartesian movements and therefore to the tool centre point (TCP) of the robot system?

Do you have any idea how I can solve the problem?

Best
matl

jrgnicho, 2017-09-07:
@mat-l  this inquiry seems to be outside the scope of this repository.  Those scaling factors are applied in the robot's joint space and as such there are unit-less. In STOMP, these scale factors are applied on the planned joint path in order to compute the time stamps, joint speeds and accelerations associated with each joint position in the path.  However, it does not directly control the speed to the TCP.

Levi-Armstrong, 2017-09-07:
You could write your own moveit adapter that which updates each point in the trajectories time to achieve a constant tool velocity, though it may not be achievable. You most likly want to do this at the planner level to ensure that this is achieved. For the Cartesian planner in this package you can write a primary cost function to achieve this.

mat-l, 2017-09-14:
Do you know if someone else is working on this issue, cause moving a robot with a defined speed along a linear path is normally a basic feature of robot controllers. I am really wondering why this is not already existing in ROS :(

jkira33, 2017-10-12:
I followed a tutorial here [https://github.com/ros-planning/moveit_tutorials/blob/kinetic-devel/doc/pr2_tutorials/planning/src/move_group_interface_tutorial.cpp#L184](url). And a comment said that setMaxVelocityScalingFactor() could reduce the speed of the robot arm via a scaling factor of the maxiumum speed of each joint. However, in my case, it did not work in tool even in joints. I have checked /joint_state in rqt, but no data in velocity category. May be something wrong in my URDF. I am confusing how to control TCP speed with move group C++ API.