ros-industrial-attic/industrial_moveit #53: STOMP failes to plan with using the the move_group move() method.

Levi-Armstrong, 2017-08-31:
It looks like the moveit move_group move method does not populate the robot start state in the planning request. Therefore STOMP check if it is populated and if not it exits and prints a warning where OMPL will plan fine. Not sure if this is really a STOMP issue or a moveit issue. Maybe if the start state is not provided it should get the current state of the robot?

janisa9, 2018-04-12:
Hi!

Is this issue solved? Recently ran into this problem and I don't know in which part is this issue.. Yesterday added comment in #37. Can you take a look?

janisa9, 2018-04-13:
@Levi-Armstrong Yep that was the problem, solved this by using move_group execute() method.

jrgnicho, 2018-07-05:
At the very least a warning should be printed when an empty robot state is encountered.