ros-industrial-attic/industrial_moveit #60: Getting started with constraint_ik

mlautman, 2017-10-10:
I'm working on a project that would benefit greatly from a planner capable of object avoidance. I'm not entirely sure where to start. As a first test, I am trying to modify the stomp_test_kr210_moveit_config demo.launch to use `kinematics_solver: constrained_ik/ConstrainedIKPlugin` as described [here].(http://wiki.ros.org/industrial_moveit/Tutorials/Adding%20constrained_ik%20to%20MoveIt%20kinematics)

Doing so results in rviz crashing with this error message: 
`ros.constrained_ik: Unable to find ros parameter: constrained_ik_solver/manipulator/constraints
ros.constrained_ik: BREAKPOINT HIT
	file = /home/mike/ws_industrial/src/industrial_moveit/constrained_ik/src/constrained_ik.cpp
	line=57

[rviz_ubuntu_64011_6174358474395917019-5] process has died [pid 64058, exit code -5, cmd /opt/ros/kinetic/lib/rviz/rviz -d 
`
This makes me think that I need to upload the constraints to the param server before this will work. 
I'm not entirely sure of the correct way to go about doing this because running the examples in the `constrained_ik/examples` directory (both as `roslaunch constrained_ik goal_tool_position_example.launch` and `roslaunch constrained_ik launch_example.launch example:=goal_tool_pointing_example use_yaml:=true`) result in the same error. 

Thanks in advance!

Levi-Armstrong, 2017-10-11:
Hello, 
The links you provided are outdated which is why things were probably are not working. [Here](http://docs.ros.org/indigo/api/constrained_ik/html/) is the updated documentation. I recommend taking a look at the [examples](http://docs.ros.org/indigo/api/constrained_ik/html/code_examples.html). There is a [yaml](http://docs.ros.org/indigo/api/constrained_ik/html/clik_config_example.html) section outline what is required for everything to work. Though the one example you mention should work so I am going to test it and see if I get the same results.

Levi-Armstrong, 2017-10-11:
So it works if you set use_yaml:=false but not when set to true. I am working on a fix now.

Levi-Armstrong, 2017-10-11:
I just created PR #61 to fix the examples when loading from yaml. Now when using the planner plugin with moveit you must replace the example_yaml with constrained_ik_solver.