ros-industrial-attic/industrial_moveit #64: Documentation on how to provide an initial trajectory ?

thibaultbarbie, 2018-01-03:
Hi,

I am using industrial moveit for my robot and the STOMP algorithm is working when I am calling it from the move_group interface in moveit. However, I would like to provide my own initial trajectory for STOMP to optimize but I cannot find how to do it.

I have found the "solve" method in stomp.cpp that takes a "Eigen::MatrixXd& initial_parameters" as parameter and I suppose I need to call it but I did not find where I should do it.

Is there some documentation about giving an initial trajectory to the STOMP algorithm ?

Levi-Armstrong, 2018-01-03:
I am not sure if it is documented anywhere currently but it looks under trajectory constraints in the planning request message for a seed trajectory seen [here](https://github.com/ros-industrial/industrial_moveit/blob/kinetic-devel/stomp_moveit/src/stomp_planner.cpp#L495).

thibaultbarbie, 2018-01-04:
Thank you for your answer! So if I understand correctly I need to call stomp from moveit with a "MotionPlanRequest" which contains my initial trajectory, is it correct ?

I checked the "MotionPlanRequest" message declared [here](http://docs.ros.org/kinetic/api/moveit_msgs/html/msg/MotionPlanRequest.html) and there is something that is unclear to me. The stomp code you gave me take the trajectory constraint as its initial trajectory, however the definition of this constraint is 
`# The constraints the resulting trajectory must satisfy
`
`TrajectoryConstraints trajectory_constraints `

So it means that instead of the resulting trajectory the stomp algorithm change it into the initial one ?

Levi-Armstrong, 2018-01-04:
>So if I understand correctly I need to call stomp from moveit with a "MotionPlanRequest" which contains my initial trajectory, is it correct ?

It is not required. You can use the [planner](https://github.com/ros-industrial/industrial_moveit/blob/kinetic-devel/stomp_moveit/src/stomp_planner.cpp) directly in your code. I would also recommend running [rosdoc_lite](http://wiki.ros.org/rosdoc_lite) to generate the documentation for this repository because is has several examples and documentation on how to configure stomp.

Note: Also, you can a further step back and use the stomp_core package directly and use the stomp_moveit package as a guide for setting things up if you don't want to use moveit messages and use your own.

>Therefore we cannot use it to specify an initial trajectory (as my trajectories will vary through time). Did I miss something ?

Your assessment of the intended use of TrajectoryConstraints is accurate, but currently it is not used or checked in MoveIt. It appears to us that it was a place holder but as of now nothing has been implemented within MoveIt that uses this. So we decided to take advantage of this unused parameter in the message to pass a seed trajectory since the existing message does not support it to prevent change the message type. Now in the future this could break if MoveIt ever decides to implement features around the TrajectoryConstraint but we hope to get the message type updated to contain a place for a seed trajectory.

thibaultbarbie, 2018-01-11:
So I suceeded to give an initial trajectory through moveit. However it was far from easy because the moveit move_group interface had no method to set a trajectory constraint. Eventually I rewrote the moveit source code to add one.

I tried to use the planner directly but the examples given were not simple enough to help me understand how to use it on my user case sorry. 

Thank you for your help! I will try to write some documentation about giving an initial trajectory to STOMP in case of other people would like to do it also.

commit 006e8c81 (2018-09-10): Merge pull request #73 from Jmeyer1292/docs/issue64_seed_setting

Discussion on Seed Trajectories "fixes #64"
closed 2018-09-10 by jrgnicho