ros-industrial-attic/industrial_moveit #66: Using OMPL and STOMP at the same time

yijiangh, 2018-03-03:
I'm wondering if we can load OMPL's planner, if STOMP fails to find a solution? In current tutorials, it seems like we can only specify one in the planning_pipeline configuration.

Thanks!

gavanderhoorn, 2018-03-03:
This is a limitation currently in MoveIt. It only loads a single planner plugin.

It's technically possible to load multiple, but it just hasn't been implemented.

Levi-Armstrong, 2018-03-03:
What we have done in the past is create a custom planning plugin which loads the stomp plugin and ompl plugin to accomplish what you want.

yijiangh, 2018-03-03:
@Levi-Armstrong So do we need to have two separated move group to load these two plugins?

Levi-Armstrong, 2018-03-04:
You should not.

Levi-Armstrong, 2018-03-04:
Sorry just read this again, when you refer to move group is that robot move groups or move group nodes?

yijiangh, 2018-03-04:
@Levi-Armstrong  Move group nodes.

Levi-Armstrong, 2018-03-04:
If you create a custom plugin that incorporates the two then you should not need two move group nodes.

yijiangh, 2018-03-04:
@Levi-Armstrong Just to make sure I understand it correctly: 

I need to create a customized `planning_interface` plugin like we did in [stomp_moveit](https://github.com/ros-industrial/industrial_moveit/blob/kinetic-devel/stomp_moveit/include/stomp_moveit/stomp_planner_manager.h) and incorporate functions to call different planners, so we can tell the plugin which planner it should use (e.g. through a `SetPlanner(string)` function) when it takes `moveit_msgs::MotionPlanRequest`? Am I right?

Levi-Armstrong, 2018-03-05:
That is correct.

yijiangh, 2018-03-05:
Thanks for your feedback @Levi-Armstrong @gavanderhoorn . 

Then I think it's proper to close this issue since the resolution is not directly related to STOMP.

closed 2018-03-05 by yijiangh

gavanderhoorn, 2018-03-05:
I've implemented something similar to what @Levi-Armstrong describes some time ago. I'll see if I can get that in a state that I can push it somewhere.

It's not that exciting really: create a planner plugin taht essentially loads other planning plugins and then implement *some way* for moveit clients to indicate which planning plugin they would like to use.

I went with a plugin prefix, so planning requests would specify something like `ompl/SBLkConfigDefault`. The multi-planner plugin planning plugin (yeah ..) figures out the prefix and forwards the request to whichever planning plugin was registered for that prefix.

yijiangh, 2018-03-05:
@gavanderhoorn 

> I've implemented something similar to what @Levi-Armstrong describes some time ago. I'll see if I can get that in a state that I can push it somewhere.

It'd be fantastic if you could share your previous work!

> I went with a plugin prefix, so planning requests would specify something like ompl/SBLkConfigDefault. The multi-planner plugin planning plugin (yeah ..) figures out the prefix and forwards the request to whichever planning plugin was registered for that prefix.

I agree. Specifying something like `primary planning plugin` and `secondary planning plugin` in a config file. Then the multi-planner plugin loads them and forwards the request accordingly.

gavanderhoorn, 2018-03-05:
@yijiangh: see [gavanderhoorn/moveit_mpp](https://github.com/gavanderhoorn/moveit_mpp).

yijiangh, 2018-03-06:
@gavanderhoorn Thank you so much! Really appreciate it.