ros-industrial-attic/industrial_moveit #70: Compilation error industrial_collision_detection

Souphis, 2018-06-19:
Hello, all. I want to use STOMP with my UR5 but I cannot compile industrial_collision_detection. I've tried compiled it on many machines (virtual os etc.) and always received following error: 
`error: redefinition of ‘struct collision_detection::DistanceRequest’ struct DistanceRequest` 
and of course many more.

I thought that maybe it was problem with Boost 1.66, however even on clean Ubuntu 16.04 with Boost 1.58 it doesn't work. I would be grateful for any advises.

My setup:
- Ubuntu 16.04
- ROS Kinetic
- Packages for UR5 and Robotiq gripper (and many more ofc).

ipa-mjp, 2018-06-19:
Hello,
I am getting same problem. Did you solved your error?

Souphis, 2018-06-19:
Not yet. However I've tried compile it on ROS-I Industrial Training VM and to my surprise, everything works well there. Maybe some dependencies are missing or sth.

Souphis, 2018-06-19:
I got same errors on my PC. I compared ros packages installed on ROS-I VM and my Ubuntu, and there is not difference besides some additionals.

ipa-mjp, 2018-06-19:
Hello,
I don't have any idea what am doing wrong. Do you have any idea to resolve this error? 
Thank you

trent-weiss, 2018-06-19:
I am getting an identical error.

It appears the issue is related to redeclaring a struct collision_detection::DistanceRequest that already exists in MoveIt.

Not sure how to fix it.

raghavendersahdev, 2018-06-19:
Hi @trent-weiss  @ipa-mjp  @Souphis .  I had the exact same issue but I was able to build and run STOMP successfully after doing the following. I had to remove 3 packages. You need to remove the following folders / packages from industrial_move_it:
- _industrial_collision_detection_: Recently everything that was in industrial_collision_detection package was merged into moveit (reason for multiple definitions) adding the ability to get detailed distance information. You will need to remove this package from the repository and try and build again. Also remove industrial_collision_detection from any CMakeLists.txt . credits to @Levi-Armstrong  for pointing me to this.
- _constraint_ik_: package / folder as it had includes from the industrial_collision_detection package like [this](https://github.com/ros-industrial/industrial_moveit/blob/kinetic-devel/constrained_ik/include/constrained_ik/solver_state.h#L34), [this](https://github.com/ros-industrial/industrial_moveit/blob/kinetic-devel/constrained_ik/src/constrained_ik.cpp#L243), [this](https://github.com/ros-industrial/industrial_moveit/blob/kinetic-devel/constrained_ik/include/constrained_ik/constraints/avoid_obstacles.h#L32), etc
- _industrial_moveit_benchmarking_: since this package has dependencies on the header files/classes obtained from the _constraint_ik_ package, I removed this too.

After doing the above steps, a `catkin build` or `catkin_make` should be able to fix and compile it.

ipa-mjp, 2018-06-21:
Hello @raghavendersahdev ,

Thank you for your information. I try to follow step as you mention. But I have other errors that's obvious. The reason is
- constraint_ik: include/constrained_ik/constraints /avoid_obstacles.h where they access function of industrial_collision_detection class while that functions/structs are not part of MoveIt! at this moment. I don't think so I can remove whole industrial_collision_detection repo. (I just mentioned one error might be possible form other packages and/or files)

- If you already did this and compile and/or build fine. Than can you give me info/access of your repo? So that I can try in my pc. Did you create PR after solving this error?

- If any one solve this problem please suggest me.
Thank you.

raghavendersahdev, 2018-06-21:
Hi @ipa-mjp as per my previous comment, you need to remove 3 all packages not just the _industrial_collision_detection_ package. The 3 packages/folder you need to remove are:

- _industrial_collision_detection_
- _constraint_ik_
- _industrial_moveit_benchmarking_

After removing these 3 packages completely, your code should successfully build....If there is a dependency of any of these packages listed in any CMakeLists.txt file, please remove that too.

ipa-mjp, 2018-06-22:
Hello @raghavendersahdev ,

Thank you so much for your help. Now after removing all 3 packages, I am able to compile and build.

jrgnicho, 2018-07-05:
@ipa-mjp  by removing you mean deleting from the package.xml list or deleting the actual packages themselves?

raghavendersahdev, 2018-07-05:
@jrgnicho  both worked for me:
- completely removing / deleteing the 3 packages also works, or Alternatively
- simply removing from package.xml and using CATKIN_IGNORE for the three packages also  worked

I removed / deleted industrial moveit and did catkin_ignore for the 2 packages (constriaint_ik and industrial_moveit_benchmarking). I did this as in my [commit](https://github.com/ros-industrial/industrial_moveit/commit/943f6989997babfd68f6d51ecd4976b7b31b4b3e)

ipa-mjp, 2018-07-06:
Hi @jrgnicho,

 Even both worked for me.

Ridhwanluthra, 2018-07-11:
@raghavendersahdev if those packages are removed it will not be able to use those functionalities.
@ipa-hsd mentioned that [this commit](https://github.com/ros-planning/moveit/commit/f235eac6d4004c3990a5edaad892076f96fd6e77#diff-867999ebefdd96976952c148a508517a) in moveit leads to the error.
Is it that the functionality from here was shifted to moveit? coz they seem similar.

closed 2018-10-10 by Souphis