ros-industrial-attic/industrial_moveit #90: stomp planner (Moveit/Rviz) no library loading in my package

khalilibrahim, 2019-04-11:
I have problem when I install stomp planner (Moveit/Rviz) no library loading in my package.
As you see in the following pictures"
[image]
[image]

gavanderhoorn, 2019-04-11:
Please do not post screenshots of text. The terminal and your editor are all text, no need for images. Just copy-paste it into your issue.

gavanderhoorn, 2019-04-11:
Also: you must provide much more information if you'd like us to assist you: Why are you still using Indigo? How did you install everything? Which packages did you install? If built from sources: which versions/branches?

Without such information we cannot help you.

khalilibrahim, 2019-04-11:
I'm so sorry.
I didn't make it again.

Thank you.
Best regards,

Khalil Ibrahim, PhD
Mechanical Engineering Department
Faculty of Engineering, Assiut University
=========================
Address: Mechanical engineering department,Faculty engineering, Assiut University.
E-mail: khalil.ibrahim77@gmail.com,  khalil.ibrahim@aun.edu.eg.
Mobile:  +201155977599, +201147014783.
Tel.Internal: 1785.
Fax: +2-2335572.

khalilibrahim, 2019-04-11:
Asi told you I follow all instruction off this wesite to use STOMP https://github.com/ros-industrial/industrial_moveit
[https://avatars1.githubusercontent.com/u/2932528?s=400&v=4]<https://github.com/ros-industrial/industrial_moveit>

GitHub - ros-industrial/industrial_moveit: ROS-Industrial movit meta-package (http://wiki.ros.org/industrial_moveit) - The world's leading software development platform · GitHub<https://github.com/ros-industrial/industrial_moveit>
Join GitHub today. GitHub is home to over 31 million developers working together to host and review code, manage projects, and build software together.
github.com

But I fail to do that if can you help.

Thank you.
Best regards,

Khalil Ibrahim, PhD
Mechanical Engineering Department
Faculty of Engineering, Assiut University
=========================
Address: Mechanical engineering department,Faculty engineering, Assiut University.
E-mail: khalil.ibrahim77@gmail.com,  khalil.ibrahim@aun.edu.eg.
Mobile:  +201155977599, +201147014783.
Tel.Internal: 1785.
Fax: +2-2335572.

gavanderhoorn, 2019-04-11:
We recently merged #86, so unless you changed branches (to `indigo-devel` or `kinetic-devel` fi), those instructions are not going to work.

Hence my question: which commands did you use exactly?

khalilibrahim, 2019-04-11:
I didn't understand what do you mean exactly ?

But my question, I want to use STOMP planner in my package motoman_sda10f. How I can install it in indigo? also I used Ubuntu 14.04?
Can you help me?

Thank you.
Best regards,

Khalil Ibrahim, PhD
Mechanical Engineering Department
Faculty of Engineering, Assiut University
=========================
Address: Mechanical engineering department,Faculty engineering, Assiut University.
E-mail: khalil.ibrahim77@gmail.com,  khalil.ibrahim@aun.edu.eg.
Mobile:  +201155977599, +201147014783.
Tel.Internal: 1785.
Fax: +2-2335572.

spy14414, 2019-05-10:
Hi,  I have the same error when I use stomp planner.
Have you solved this problem? Can you help me?
Thanks in advance.