ros-industrial-attic/industrial_moveit #103: Building error with Melodic

lepalom, 2020-05-29:
I'm trying to build it with melodic and I have found this building error:

```
 Building CXX object CMakeFiles/industrial_collision_detection_plugin.dir/src/industrial_collision_detection_plugin.cpp.o
In file included from /usr/ros/melodic/include/moveit/planning_scene/planning_scene.h:43,
                 from /usr/ros/melodic/include/moveit/collision_detection/collision_plugin.h:41,
                 from /srv/robotica/ros/src/ros-industrial/industrial_moveit/industrial_collision_detection/include/industrial_collision_detection/industrial_collision_detection_plugin.h:28,
                 from /srv/robotica/ros/src/ros-industrial/industrial_moveit/industrial_collision_detection/src/industrial_collision_detection_plugin.cpp:26:
/usr/ros/melodic/include/moveit/collision_detection/collision_detector_allocator.h: In instantiation of ‘const string& collision_detection::CollisionDetectorAllocatorTemplate<CollisionWorldType, CollisionRobotType, CollisionDetectorAllocatorType>::getName() const [with CollisionWorldType = collision_detection::CollisionWorldIndustrial; CollisionRobotType = collision_detection::CollisionRobotIndustrial; CollisionDetectorAllocatorType = collision_detection::CollisionDetectorAllocatorIndustrial; std::__cxx11::string = std::__cxx11::basic_string<char>]’:
/usr/ros/melodic/include/moveit/collision_detection/collision_detector_allocator.h:79:22:   required from here
/usr/ros/melodic/include/moveit/collision_detection/collision_detector_allocator.h:81:44: error: ‘NAME’ is not a member of ‘collision_detection::CollisionDetectorAllocatorIndustrial’
     return CollisionDetectorAllocatorType::NAME;
                                            ^~~~
make[2]: *** [CMakeFiles/industrial_collision_detection_plugin.dir/build.make:63: CMakeFiles/industrial_collision_detection_plugin.dir/src/industrial_collision_detection_plugin.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:73: CMakeFiles/industrial_collision_detection_plugin.dir/all] Error 2
make: *** [Makefile:141: all] Error 2

```

any idea?

jrgnicho, 2020-05-29:
Looks like a change in moveit broke this.  The CollisionDetectorAllocatorIndustrial inherits from a base class in moveit and it appears that the NAME variable has been removed

hoanggiang84, 2020-10-02:
I changed to NAME_ in the file collision_detector_allocator.h.