ros-industrial-attic/industrial_moveit #104: melodic code no longger build

mpatalberta, 2021-04-07:
error_message<<".\nSet or add the 'collision_detector' parameter to an allowed collision detector in the move_group.launch file"<<std::endl;
        throw std::runtime_error(error_message.str());
      }
    }

    state.collision_robot = planning_scene->getCollisionRobot();

    state.collision_world = planning_scene->getCollisionWorld();
  }

  }

The object getCollisionRobot and getCollisionWorld no longer exist

mpatalberta, 2021-04-07:
/moveitmaster_ws/src/industrial_moveit/constrained_ik/src/constrained_ik.cpp

mpatalberta, 2021-04-07:
At about line 247

Levi-Armstrong, 2021-04-08:
Most likely MoveIt change something in the API which has broken things. This package is largely unmaintained since all of the capability is now provided within [trajopt_ros](https://github.com/ros-industrial-consortium/trajopt_ros) repository. If you create a PR which fixes the issue I would be happy to review and merge.